rodneybakiskan/ps5-esp32 #1: Won't connect to Dualsense Controller

mbackas, 2022-01-30:
Hi, having trouble getting ps5Connect example sketch to work. ps5Controller.begin("mac address") doesn't work.

rodneybakiskan, 2022-02-01:
Make sure you've previously connected the controller to another device you know the bluetooth mac address of. Use this mac address. It might take a few tries for the ps5 controller to connect. Let me know if it works!

mbackas, 2022-02-04:
Thanks for the advice. What's weird is I'm able to pair a PS3 controller using the ps3-esp32 library linked in the README.md, but I'm not able to pair a ps5 controller using this library.

To pair, I've first paired both controllers to my phone, and used my phone's bluetooth MAC address as the argument for the begin() function.

I just ordered some new ESP-32S dev boards with the ESP32-WROOM modules and I'll try again and let you know, thanks for your help!

mbackas, 2022-02-07:
Fixed! Tried again on another board. You're right, controller has to be connected to computer/phone first, then input the computer/phone Bluetooth MAC address into the arduino sketch. Note I had to change the begin() function to accept uint8_t* instead of unsigned char*. Then behin() simply passes the uint8_t* to the enclosed bluetooth function.

closed 2022-02-07 by mbackas

gigahidjrikaaa, 2022-12-01:
@mbackas could you share your modified codes? I think I am having the same problem as you did. Thanks!