rodneybakiskan/ps5-esp32 #4: Error when compiling

Giesenhausener, 2022-06-05:
I followed your instructions and get this error when compiling:

/Users/dirk/Documents/Arduino/libraries/ps5-esp32-main/src/ps5_spp.c: In function 'sppCallback':
/Users/dirk/Documents/Arduino/libraries/ps5-esp32-main/src/ps5_spp.c:89:30: error: 'ESP_BT_SCAN_MODE_CONNECTABLE' undeclared (first use in this function); did you mean 'ESP_BT_NON_CONNECTABLE'?
     esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE);
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                              ESP_BT_NON_CONNECTABLE
/Users/dirk/Documents/Arduino/libraries/ps5-esp32-main/src/ps5_spp.c:89:30: note: each undeclared identifier is reported only once for each function it appears in
/Users/dirk/Documents/Arduino/libraries/ps5-esp32-main/src/ps5_spp.c:89:5: error: too few arguments to function 'esp_bt_gap_set_scan_mode'
     esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE);
     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/dirk/Documents/Arduino/libraries/ps5-esp32-main/src/ps5_spp.c:4:
/Users/dirk/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api/esp_gap_bt_api.h:479:11: note: declared here
 esp_err_t esp_bt_gap_set_scan_mode(esp_bt_connection_mode_t c_mode, esp_bt_discovery_mode_t d_mode);
           ^~~~~~~~~~~~~~~~~~~~~~~~
exit status 1

Can you see what I missed?

Giesenhausener, 2022-06-05:
I removed the comments in line 86-90 in file ps5_spp.c
After this compiling didi not throw any errors

closed 2022-06-05 by Giesenhausener

rodneybakiskan, 2022-10-19:
Thanks! I'll update the readme. It looks like an ESP-IDF compatability variable needs to be changed depending what esp version you have.