rodneybakiskan/ps5-esp32 #5: Other PS5 BT library

vlinder, 2022-08-07:
I found this other library that supports a lot of functionality of the PS5 controller. I don't have the expertise to know. Could this be "easily" ported to the esp32?

https://github.com/felis/USB_Host_Shield_2.0#ps5-library

* buttons
* joysticks
* touchpad and IMU
* rumble
* lightbar
* microphone LED
* player LEDs
* haptic trigger effects
* pairing

It is apparently based on the official Sony linux driver.

rodneybakiskan, 2022-10-19:
Did you manage try the other library out?

vlinder, 2022-10-21:
Sorry, it didn't support the built in bluetooth of the esp32 unfortunately.. :/

It should be possible to port it, but might be a lot of work, probably..

https://github.com/felis/USB_Host_Shield_2.0/issues/572

vlinder, 2023-08-30:
@rodneybakiskan I found this bluepad32 library that supports basically everything you ever want :D

https://github.com/ricardoquesada/bluepad32