rodneybakiskan/ps5-esp32 #7: 'ESP_BT_SCAN_MODE_CONNECTABLE' undeclared

hoimaly, 2022-09-17:
c:\Users\TrongPhuc\OneDrive - VNU-HCMUS\Documents\Arduino\libraries\ps5-esp32-main\src\ps5_spp.c: **_In function 'sppCallback':_**
c:\Users\TrongPhuc\OneDrive - VNU-HCMUS\Documents\Arduino\libraries\ps5-esp32-main\src\ps5_spp.c:89:30: error: **_'ESP_BT_SCAN_MODE_CONNECTABLE' undeclared (first use in this function); did you mean 'ESP_BT_NON_CONNECTABLE'?
     esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE);
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                              ESP_BT_NON_CONNECTABLE_**
c:\Users\TrongPhuc\OneDrive - VNU-HCMUS\Documents\Arduino\libraries\ps5-esp32-main\src\ps5_spp.c:89:30: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\TrongPhuc\OneDrive - VNU-HCMUS\Documents\Arduino\libraries\ps5-esp32-main\src\ps5_spp.c:89:5: error: **_too few arguments to function 'esp_bt_gap_set_scan_mode'
     esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE);_**
     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\TrongPhuc\OneDrive - VNU-HCMUS\Documents\Arduino\libraries\ps5-esp32-main\src\ps5_spp.c:4:
C:\Users\TrongPhuc\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.5/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api/esp_gap_bt_api.h:480:11: note: declared here
 esp_err_t esp_bt_gap_set_scan_mode(esp_bt_connection_mode_t c_mode, esp_bt_discovery_mode_t d_mode);
           ^~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

rodneybakiskan, 2022-10-19:
Seems like an ESP-IDF compatability variable needs to be changed depending what esp version you have. I'll update the readme accordingly. Thanks!

closed 2022-10-19 by rodneybakiskan

hoimaly, 2022-10-19:
Hi, thank you for your response and solving my problem.  Your library is awesome, I love it!