rodneybakiskan/ps5-esp32 #8: Failed to Connect Dualsense

Infinicus, 2022-11-17:
I have this problem where the Dualsense disconnects as it is trying to connect to the ESP-32. FYI, the dev. board that I am using is the ESP-32 Devkitc V4.

Is there any idea on solving this problem? Or is this a problem caused by my own Dualsense and not the ESP? Thanks.

rodneybakiskan, 2022-11-19:
It sometimes takes a few tries to connect. Make sure you've changed the MAC addresss as detailed in the README. Otherwise I can't say for sure, but it's unlikely to be a dualsense problem especially if you're able to connect it to other devices.

gigahidjrikaaa, 2022-11-23:
I tried a bunch of times, and it won't connect.  It just keeps turning off as it tries to connect to the ESP-32. Maybe I'll post a video of the problem in the future. Thanks for the help.

TANG617, 2022-12-15:
#11 Hi, I found a serious problem in README that leads to your problem!

gigahidjrikaaa, 2022-12-15:
Refer to #11 by @TANG617

closed 2022-12-15 by gigahidjrikaaa