rodneybakiskan/ps5-esp32 #10: Is ESP32-S3 available for this library?

TANG617, 2022-12-14:
While I compile for ESP32-S3, I got the following errors from ` esp_spp_init `
How to solve it? 
Thanks!
```bash
Dependency Graph
|-- Wire @ 2.0.0
|-- lvgl @ 8.3.0-dev
|-- ps5Controller @ 2.1
|-- UI
|   |-- lvgl @ 8.3.0-dev
Building in release mode
Compiling .pio/build/esp32s3/src/main.cpp.o
In file included from lib/lvgl/src/misc/../lv_conf_internal.h:39,
                 from lib/lvgl/src/misc/lv_log.h:16,
                 from lib/lvgl/lvgl.h:25,
                 from lib/UI/ui.h:15,
                 from src/main.cpp:4:
lib/lvgl/lv_conf.h:355: warning: "LV_FONT_MONTSERRAT_12" redefined
 #define LV_FONT_MONTSERRAT_12 0
 
<command-line>: note: this is the location of the previous definition
In file included from lib/lvgl/src/misc/../lv_conf_internal.h:39,
                 from lib/lvgl/src/misc/lv_log.h:16,
                 from lib/lvgl/lvgl.h:25,
                 from lib/UI/ui.h:15,
                 from src/main.cpp:4:
lib/lvgl/lv_conf.h:357: warning: "LV_FONT_MONTSERRAT_16" redefined
 #define LV_FONT_MONTSERRAT_16 0
 
<command-line>: note: this is the location of the previous definition
src/main.cpp:6: warning: "TOUCH_MODULES_CST_MUTUAL" redefined
 #define TOUCH_MODULES_CST_MUTUAL
 
<command-line>: note: this is the location of the previous definition
Linking .pio/build/esp32s3/firmware.elf
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o):(.literal.ps5_l2cap_disconnect_ind_cback+0x4): undefined reference to `L2CA_DisconnectRsp'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o):(.literal.ps5_l2cap_config_ind_cback+0x0): undefined reference to `L2CA_ConfigRsp'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o):(.literal.ps5_l2cap_connect_ind_cback+0x8): undefined reference to `L2CA_ErtmConnectRsp'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o):(.literal.ps5_l2cap_connect_ind_cback+0xc): undefined reference to `L2CA_ConfigReq'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o):(.literal.ps5_l2cap_init_service+0x14): undefined reference to `L2CA_Register'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o):(.literal.ps5_l2cap_reconnect+0x8): undefined reference to `L2CA_ErtmConnectReq'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o):(.literal.ps5_l2cap_send_hid+0x14): undefined reference to `L2CA_DataWrite'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o): in function `ps5_l2cap_disconnect_ind_cback':
/Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_l2cap.c:319: undefined reference to `L2CA_DisconnectRsp'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o): in function `ps5_l2cap_config_ind_cback':
/Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_l2cap.c:301: undefined reference to `L2CA_ConfigRsp'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o): in function `ps5_l2cap_connect_ind_cback':
/Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_l2cap.c:232: undefined reference to `L2CA_ErtmConnectRsp'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_l2cap.c:235: undefined reference to `L2CA_ErtmConnectRsp'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_l2cap.c:238: undefined reference to `L2CA_ConfigReq'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o): in function `ps5_l2cap_init_service':
/Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_l2cap.c:189: undefined reference to `L2CA_Register'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o): in function `ps5_l2cap_reconnect':
/Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_l2cap.c:119: undefined reference to `L2CA_ErtmConnectReq'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_l2cap.c.o): in function `ps5_l2cap_send_hid':
/Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_l2cap.c:159: undefined reference to `L2CA_DataWrite'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_spp.c.o):(.literal.sppCallback+0x0): undefined reference to `esp_bt_gap_set_scan_mode'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_spp.c.o):(.literal.sppInit+0x14): undefined reference to `esp_spp_register_callback'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_spp.c.o):(.literal.sppInit+0x18): undefined reference to `esp_spp_init'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_spp.c.o): in function `sppCallback':
/Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_spp.c:86: undefined reference to `esp_bt_gap_set_scan_mode'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/lib291/libps5-esp32.a(ps5_spp.c.o): in function `sppInit':
/Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_spp.c:56: undefined reference to `esp_spp_register_callback'
/Users/timli/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/timli/Documents/PlatformIO/Projects/DS32S3/lib/ps5-esp32/src/ps5_spp.c:65: undefined reference to `esp_spp_init'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32s3/firmware.elf] Error 1
===================== [FAILED] Took 7.02 seconds =====================

```

rodneybakiskan, 2023-01-14:
Not sure, sorry. But my best guess is that there is some incompatability with the ESP-IDF. All I can say is to make sure everthing is up to date.

GPT0310, 2023-02-02:
I think the problem is that the PS5 controller only works with Classic Bluetooth and if I'm not mistaken the ESP32-S3 board only has BLE, which are incompatible.

closed 2023-02-19 by rodneybakiskan