rodneybakiskan/ps5-esp32 #11: The Doc is wrong, and the Address in bracket should be the MAC Address of Controller!

TANG617, 2022-12-15:
I tried out by myself.
The README on GitHub is basically wrong!
The ESP32 would automatically connect the PS5 Controller with the Bluetooth MAC Address in `ps5.begin()` bracket.
So, the MAC Address should be the address of PS5 Controller, not the MAC Address of "Controller connected console/PC"(That's where the doc went wrong!)
[image]
Take the above as an example, the MAC address filled should be like ` ps5.begin("bc:c7:46:33:11:d2");`, **NOT** `ps5.begin("f8:4d:89:66:16:9d"); ` as said in README

gigahidjrikaaa, 2022-12-15:
I got notified of your issue. Thanks for this, I'll try it out and let you know after this.

gigahidjrikaaa, 2022-12-15:
Wow, it actually worked.
I uploaded the code and get my Dualsense to pairing mode after turning on the ESP-32. Then, it finally connects! Nice solution!

nonobody15, 2022-12-16:
**Thank you @ TANG617!**
long story short: a month ago I already used my PS5 controller with the library and I connected to my iPad as it says in the README, then entered the iPad MAC address in the setup of the ps5ReceiveData and it just worked without any problems! I was very impressed.

BUT! a few days ago I restarted the ps55 controller project and ... nothing worked, the controller just wouldn't connect to the ESP32, I was going crazy. I tried to rule out all possible error causes. But, nothing!
Yesterday I read the post and immediately tried it out. **It works!**

If anyone has any idea why the method to connect via BLE has changed, please share it with me. I would be very pleased.

rodneybakiskan, 2022-12-16:
Thanks @TANG617 for pointing that out. I'll make sure to update the README accordingly.

@nonobody15  The BLE method changed due to a recent update by @asukiaaa, which made connecting the ps5 controller much more reliable. Previously, the esp32 connected by mimicking the device the ps5 controller was looking to connect to (which is why you needed the bluetooth address of your console/phone). The updated code initiates a handshake with the ps5 controller before connecting, hence it's now the address of the ps5 controller you need.

closed 2022-12-16 by rodneybakiskan

asukiaaa, 2022-12-16:
Thank you for updating the readme.
By the way, ps5 controller is bluetooth classic. Not BLE.