rodneybakiskan/ps5-esp32 #12: E (6487) ps5_L2CAP: L2CA_CONNECT_REQ ret=64

fabriofabrio, 2023-01-08:
Hi

I have installed your library 

My controller doesn't connect to esp32 dev kit

this is the log

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (14) ps5_L2CAP: L2CA_CONNECT_REQ ret=0

Ready.
E (6487) ps5_L2CAP: L2CA_CONNECT_REQ ret=64

E (12487) ps5_L2CAP: L2CA_CONNECT_REQ ret=65

E (18488) ps5_L2CAP: L2CA_CONNECT_REQ ret=64

E (24489) ps5_L2CAP: L2CA_CONNECT_REQ ret=65

I serced abouth L2CA_CONNECT_REQ on internet, and library, but nothing

tanx for your time

atij, 2023-01-08:
same here, not working with ESP32-D0WD-V3 (revision v3.0)

fabriofabrio, 2023-01-09:
Thank you for Quick reply

rodneybakiskan, 2023-01-13:
Have you pressed and held the PS button + Create button until the controller goes into paring mode? It'll start flashing with 2 pulses of blue at a time.

fabriofabrio, 2023-01-14:
Thank you for reply

Yes I have follow the correct procedure, but nothing connection

rodneybakiskan, 2023-01-14:
I know it works with the DOIT devkit containing the ESP32-WROOM-32 module (which uses the ESP32-D0WD-V3 core mentioned by @atij)

atij, 2023-01-14:
did not manage to make it work with doit devkit, it works with wemos D1 R32 though

fabriofabrio, 2023-01-14:
It Is strange: ESP32-D0WD-V3 have bt 4.1 edr that Is retro compatible with bt 2.1 edr

Thank you very much

closed 2023-02-19 by rodneybakiskan