rodneybakiskan/ps5-esp32 #14: Pairing routine help

rdavies02, 2023-02-17:
Is there a way to put the ps5 controller in pairing mode and have it connect to the esp with this library? 

I want to use multiple different controllers with a single esp (but not all at the same time) and am investigating a more efficient method of pairing without having to go the "phone route".

I am about to look into using the ```esp_gap_bt_api.h``` but I have no idea where in the code to put the function calls

any help would be much appreciated :)

rodneybakiskan, 2023-02-19:
I have never tried to use multiple ps5 controllers. 

With that being said, you could try calling ps5.begin("_mac address of target controller_") during runtime whenever you want to switch to a different controller (I have never tested this). You would still need to determine the mac address of each controller though.

rdavies02, 2023-03-14:
Turns out I wasn't using an updated version of the library :P thanks for the help though.

We were able to create a pairing routine, if you are interested here is a link to our code:

https://github.com/PolarRobotics/ESP32PRCodebase/blob/IntegratingSpecialBots2ESP/src/pairing.cpp

closed 2023-03-14 by rdavies02