rodneybakiskan/ps5-esp32 #15: Help pairing!

jb2057, 2023-02-19:
The PS5 controller pairing is ok using the Ps3Controller.h or PS4Controller.h, but not with ps5Controller.h

I tried:
- two original Sony PS5 Controllers with the latest firmware
- two ESP32 boards - KeeYees ESP32S and DOIT DEVIT V1
- two ESP32 core versions - 1.0.6 and 2.0.6

The SixaxisPairTool cannot recognize PS5 - only PS4 and PS3, so I cannot try it.

Are there specific boards tested for connectivity with PS5 controller?

jb2057, 2023-02-20:
Yes, I am getting the MAC from my PC and phone. 
Any ideas for SixaxisPairTool alternative to write the MAC?

jb2057, 2023-02-20:
Just tried a 3rd board - HiLetgo ESP-WROOM-32.
No luck.
Anyone can recommend specific board that works?

jb2057, 2023-02-28:
After testing with 3 ESP32 boards and 6 PS controllers (PS3, PS4 and 3x PS5) - I am 100% sure the issue is the new firmware of PS5.
PS5 with the last firmware 0307 will NOT connect.
PS5 with 0282 will connect.
Somehow the new PS5 firmware and the current PS5 library are not understanding each other...

rodneybakiskan, 2023-03-02:
You could try using an older version of the esp32-library before the changes asukiaaa kindly made. It's possible that may work. Note that you would need to use the bluetooth MAC address of **your phone or console** rather than the controller (as detailed in the previous readme).

rodneybakiskan, 2023-03-02:
Otherwise reverting to an older firmware on the controller is always an option

amilisov, 2023-03-13:
Is this issue fixed?
Also according to the license agreement of the controller https://www.playstation.com/en-us/legal/eula-firmware-updater/
You can't legaly downgrade :)

rodneybakiskan, 2023-03-14:
@amilisov Do you have this connectivity issue as well?

amilisov, 2023-03-16:
Okay it works now. I erased the flash of the esp32 and ran the ps5Connect example again and magicaly everythink is okay now. https://randomnerdtutorials.com/esp32-erase-flash-memory/

jb2057, 2023-03-16:
Can you please tell us the firmware version of your controller? I believe the problem is that - not the esp32Sent from my Galaxy

amilisov, 2023-03-16:
Firmware version 0356

mrkacperso, 2023-06-28:
> You could try using an older version of the esp32-library before the changes asukiaaa kindly made. It's possible that may work. Note that you would need to use the bluetooth MAC address of **your phone or console** rather than the controller (as detailed in the previous readme).

Hi I had the same issue and went with your suggestion to try older version of the library which sets `ps5SetBluetoothMacAddress(addr)` in [ `ps5Controller.cpp`](https://github.com/rodneybakiskan/ps5-esp32/commit/1264d0b65272b4bda9df0b9085a23aeac2f5e9c9#diff-782c965b1bb522cc2f6f095a3010b5a83fe129c01023c9343e68913cba405c21) and it works like a charm, even with firmware version 0356.
Just out of curiosity, why was this change introduced?

jb2057, 2023-06-28:
How to get this older version?Sent from my Galaxy

jb2057, 2023-06-28:
YES!!! The old version of ps5Controller.cpp works. Thank you!

pdumais, 2024-01-08:
I had the same issue and was able to fix it by using the controller's MAC address (not IP address as per the README) but you also have to start the controller in pair mode first to get it working