rodneybakiskan/ps5-esp32 #16: Trouble pairing PS5 controller.

madbunnylabs, 2023-03-14:
I'm using a M5 stamp Pico board but I can't seem to get the controller to connect. 

I'm using ESP32 version 1.0.6

If I use the PS4 controller library with the "PS5" controller it pairs fine to the m5 stamp, but of course I can only use the joysticks as none of the other buttons work with the ps4 library.   Since I can connect with the PS4 library, I believe this means I have the correct mac address.    

However If I try to connect the ps5 controller to the PS5 library, the controller doesn't connect and turns off after about 20 seconds.

I get this chain of connection requests on the serial monitor

Ready
E (165032) ps5_L2CAP: L2CA_CONNECT_REQ ret=64
to
E (170033) ps5_L2CAP: L2CA_CONNECT_REQ ret=76

I've been stuck on this for the past week, so any help would be greatly appreciated.   Just don't know what else to try. 

PS:  Thank you so much for putting this library together for the esp32!

rodneybakiskan, 2023-03-15:
Have you used the MAC address of the _controller_? This is different to that of the ps4 library were you would need to use the pc/phone/console MAC address

madbunnylabs, 2023-03-16:
Thank you for the reply.  I tried using the MAC address of the controller first but wasn't having luck.  I was able to get it to work using the MAC address of the controller as you suggested and placing the PS5 controller in pairing mode for the first time as I saw you comment in one of the older posts.     Thanks for the help, I was finally able to get a small RC car working. ^^

rodneybakiskan, 2023-03-17:
I'm glad you got it working!

closed 2023-03-17 by rodneybakiskan