rodneybakiskan/ps5-esp32 #26: compilation error: exit status 1

TheOGSeansharky, 2024-07-13:
hi i am having a problem with uploading it to my board i am just trying to run the 

`
#include <ps5Controller.h>
void setup()
{
    ps5.begin("1a:2b:3c:01:01:01");
    Serial.println("Ready.");
}

void loop()
{}`

and i get 
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o:(.literal.ps5_l2cap_init_service+0x14): undefined reference to `L2CA_Register'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o:(.literal.ps5_l2cap_disconnect_ind_cback+0x4): undefined reference to `L2CA_DisconnectRsp'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o:(.literal.ps5_l2cap_config_ind_cback+0x0): undefined reference to `L2CA_ConfigRsp'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o:(.literal.ps5_l2cap_connect_ind_cback+0x8): undefined reference to `L2CA_ErtmConnectRsp'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o:(.literal.ps5_l2cap_connect_ind_cback+0xc): undefined reference to `L2CA_ConfigReq'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o:(.literal.ps5_l2cap_reconnect+0x8): undefined reference to `L2CA_ErtmConnectReq'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o:(.literal.ps5_l2cap_send_hid+0x14): undefined reference to `L2CA_DataWrite'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o: in function `ps5_l2cap_init_service':
c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_l2cap.c:189: undefined reference to `L2CA_Register'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o: in function `ps5_l2cap_disconnect_ind_cback':
c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_l2cap.c:319: undefined reference to `L2CA_DisconnectRsp'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o: in function `ps5_l2cap_config_ind_cback':
c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_l2cap.c:301: undefined reference to `L2CA_ConfigRsp'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o: in function `ps5_l2cap_connect_ind_cback':
c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_l2cap.c:232: undefined reference to `L2CA_ErtmConnectRsp'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_l2cap.c:235: undefined reference to `L2CA_ErtmConnectRsp'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_l2cap.c:238: undefined reference to `L2CA_ConfigReq'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o: in function `ps5_l2cap_reconnect':
c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_l2cap.c:119: undefined reference to `L2CA_ErtmConnectReq'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_l2cap.c.o: in function `ps5_l2cap_send_hid':
c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_l2cap.c:159: undefined reference to `L2CA_DataWrite'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_spp.c.o:(.literal.sppCallback+0x0): undefined reference to `esp_bt_gap_set_scan_mode'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_spp.c.o:(.literal.sppInit+0x14): undefined reference to `esp_spp_register_callback'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_spp.c.o:(.literal.sppInit+0x18): undefined reference to `esp_spp_init'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_spp.c.o: in function `sppCallback':
c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_spp.c:86: undefined reference to `esp_bt_gap_set_scan_mode'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ReplacedForPrivacy\AppData\Local\Temp\arduino\sketches\C437A8BF180FAC706F94BB0708D5F0B8\libraries\ps5Controller\ps5_spp.c.o: in function `sppInit':
c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_spp.c:56: undefined reference to `esp_spp_register_callback'
c:/users/ReplacedForPrivacy/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: c:\Users\ReplacedForPrivacy\OneDrive\Documents\Arduino\libraries\ps5Controller\src/ps5_spp.c:65: undefined reference to `esp_spp_init'
collect2.exe: error: ld returned 1 exit status

TheOGSeansharky, 2024-07-13:
i am using a esp 32 s3 wroom 1 made by espressif

HamzaYslmn, 2024-09-07:
Esp32 S3 doesn't have BT classic, just BLE